OpenRoberta/robertalab #436: NAO - hal.py - setLeds intensity

fgras-erm, 2017-03-08:
Hi,
In setLeds(), the function self.led.setIntensity() put all leds in white with a intensity. So, in hal.py, it's impossible to put a red, a dark green or a light blue color.

I propose :
 - either you keep intensity parameter but it applies a ratio on the color
 - or you replace the combox with a "Color Blocks" with at least 24 choices of color

Regards,
fgras-erm

JanisM91, 2017-03-09:
Hi,

i think it is best to remove the intensity and change the block so that the color is not a dropdown field but a color value. This way it is possible to add a predefined color block or to use a rgb block.

Regards
Janis

JanisM91, 2017-03-10:
Blocks now look like this:
[image]
In addition to the color picker it is also possible to use a RGB Block that takes a variable number for each channel (red,green,blue).

fgras-erm, 2017-03-11:
OK, it's a better idea than I thought.
I let you close this ticket when it's done and tested?

Regards,
fgras

JanisM91, 2017-03-23:
fixed with 5772445 and 2e417e6

closed 2017-03-23 by JanisM91