OpenRoberta/robertalab #438: NAO - hal.py - case in fsr and  touchsensors function

fgras-erm, 2017-03-08:
Hi,

For parameters of **fsr** or **touchsensors** functions, the generated python code create string with the first letter in upper.
In the hal.py, all is wrote in lowercase.

Furthermore, for head touchsensors, "position" variable is tested with "Front", "Middle" and "Rear" values, when it's "side" variable which have this values.

Regards,
fgras-erm

JanisM91, 2017-03-08:
Hi,

the difference in the capitalization was fixed with 32f3b37.

I will have a look into the variables for the touchsensors.

fgras-erm, 2017-03-09:
Hi,
The variables was already fixed too. Good job!

Do it miss a `return False` at the end?

Regards,
fgras

JanisM91, 2017-03-09:
`return False` added with OpenRoberta/robertalab-naoprogram@b2a9bdb

fgras-erm, 2017-03-10:
great!

closed 2017-03-10 by fgras-erm