OpenRoberta/robertalab #444: New startscreen

bjost2s, 2017-03-16:
[image]
[image]

ensonic, 2017-03-16:
Looks awesome. Only concern is the 'back' ('zurück') button. Maybe change the title on the 2nd screen to "Wähle dein Calliope (zurück zur Auswahl)".

bjost2s, 2017-03-17:
@ensonic, could you please attache an image of the menu from ev3dev, so that we can distinguish the two ev3 systems by the menu layout?

ensonic, 2017-03-17:
@bjost2s I have images here:
https://github.com/OpenRoberta/robertalab-ev3dev

closed 2017-03-21 by bjost2s