OpenRoberta/robertalab #445: drive/turn/steer blocks with speed=0 won't finish

ensonic, 2017-03-16:
I one creates a drive/turn/steer block with speed=0, those will run forever. While we already limit the speed to max-speed (e.g. if the user set speed to 500, we will cut it at 100), I am not sure if we should enforce a min-speed of 1.

At some point we should have a linter (maybe part of the program checker), that comments on the program to point out problematic blocks.

ensonic, 2017-03-20:
To make it extra complicated we launch the beeps with a shell
https://github.com/rhempel/ev3dev-lang-python/blob/develop/ev3dev/core.py#L3234

This makes it hard to kill them.

cvejoski, 2017-07-21:
We can only check if the speed of the motor is explicitly set by math const, if we have expression or sensor values that is not possible to be done.

For now I will add in the checker additional condition if a user set the speed of the motor to 0 then we will pop up a worning

ensonic, 2017-07-21:
Thanks, I think the check for a static value will cover a lot of the use cases. When people use a variable, this won't help of course :)

closed 2017-09-08 by bjost2s