OpenRoberta/robertalab #448: Docker Container Image Configuration for Windows 10

WizGeek, 2017-03-17:
**AS A** robotics instructor for middle school students,
**I NEED** to deploy RobertaLab in a Docker container with Wireless and USB connections,
**WITH ISSUE**: Docker/Windows configuration details are missing, incomplete, or outdated.
- - - -
**_Sample Dockerfile:_**
`##===============##================================================`
`##  OpenRoberta  ##`
`##---------------##`
`## To build>        docker build -t shms/robertalab .`
`## To probe/debug>  docker run -it shms/robertalab bash`
`## To run>          docker run -p 1999:1999 shms/robertalab`
`##------------------------------------------------------------------`
`FROM ubuntu`
` `
`RUN apt-get clean -y && apt-get -y update && apt-get -y upgrade `
`RUN apt-get install -y git phantomjs maven`
`RUN apt-get clean -y`
` `
`RUN apt-get install -y wget unzip && \`
`    wget https://github.com/OpenRoberta/robertalab/archive/master.zip && \`
`	unzip master.zip && \`
`	rm master.zip`
` `
`RUN apt-get install -y default-jdk iputils-ping`
` `
`ENV JAVA_HOME /usr/lib/jvm/java-8-openjdk-amd64`
` `
`WORKDIR /robertalab-master/OpenRobertaParent`
`RUN mvn clean install`
` `
`EXPOSE 1999`
`WORKDIR /robertalab-master`
`CMD ["/robertalab-master/ora.sh", "--start-from-git"]`

bjost2s, 2017-03-18:
Currently we do not provide any Docker configuration. 
Please explain in detail what is not working or what is missing, then we try to help and can maybe offer a docker configuration for windows here. Thank you for your assistance.

WizGeek, 2017-03-19:
> Configuration for Windows 10 (#448)
> <author@noreply.github.com>

Excellent!If you use the Dockerfile supplied in the post, you'll find that the Roberta Lab server starts and is accessible on port 1999.  I'll put the docker configuration at the end of this email as well.What is not working is the ability to connect to an EV3 brick by wireless or USB connection. The response is an immediate popup that asks if the robot is turned on, etc.I have verified the connectivity of the EV3 brick by running OR outside of Docker, and it connects both with wireless and via USB using the USB utility.Your attention to this is greatly appreciated.- - - - - -Dockerfile- - - - - - ##===============##================================================##  OpenRoberta  ####---------------#### To build>        docker build -t shms/robertalab .## To probe/debug>  docker run -it shms/robertalab bash## To run>          docker run -p 1999:1999 shms/robertalab##------------------------------------------------------------------FROM ubuntuRUN apt-get clean -y && apt-get -y update && apt-get -y upgrade RUN apt-get install -y git phantomjs mavenRUN apt-get clean -yRUN apt-get install -y wget unzip && \    wget https://github.com/OpenRoberta/robertalab/archive/master.zip && \	unzip master.zip && \	rm master.zipRUN apt-get install -y default-jdk iputils-pingENV JAVA_HOME /usr/lib/jvm/java-8-openjdk-amd64WORKDIR /robertalab-master/OpenRobertaParentRUN mvn clean installEXPOSE 1999WORKDIR /robertalab-masterCMD ["/robertalab-master/ora.sh", "--start-from-git"]

bjost2s, 2017-03-20:
I tested your dockerfile, works :-)
to make it availabe from outside (my smartphone) I had to change some settings in the firewall of my windows 10 system:
[image]
for both Container and for DockerSmbMount I changed the Protokolltype to all and the lokale port to all,
after that I could connect. Maybe your don't have to change all 3 settings.

I will test it with a robot tomorrow.

bjost2s, 2017-03-21:
maybe my o.a. post is not relevant. Today I tested it again. Please try 
run -p 1999:1999 dockerfile
to make sure your port is forwarded

WizGeek, 2017-03-22:
Based upon the Dockerfile I supplied, this is the command I issue to start the container:    docker run -p 1999:1999 shms/robertalabI run the command above inside a Windows cmd.exe command shell with Administrator privileges.In the Chrome browser, localhost:1999 gets me to the NEPO editor, and I click the EV3 top link, then connect...After getting the brick to a the registration token, and then typing that into the browser session and licking OK,I get an immediate pop-up:AttentionCheck if your robot is switched on andconnected to the server.If you have problems to do this, pleasehave a look at out help pages.[OK]1.  What can I do to enable browser/Windows/PC diagnostics to see what's being attempted and failing?2.  I have Avast anti-virus disabled, so that's not the issue.3.  Other ideas?

bjost2s, 2017-03-22:
Please check to which server you have connected the robot. By default it connects to lab.open-roberta.org. You have to tell the robot in the robot's menu the ip and port from your local machine, so that the robot connect to your server. If you cannot type in a different server ip, please enable this feature in the robot's system menu.

bjost2s, 2017-04-05:
Is it working now?

WizGeek, 2017-04-06:
I've not been successful yet.  I've tried several different combinations of IP and port on the brick, but I get errors on both the brick and OR:Brick:  Open Roberta Lab  is unreachable!    Reason:  No access from  this network!   (Press any key)OR in Chrome Browser:  Attention  Check if your robot is switched on and  connected to the server.  If you have problems to do this, please  have a look at our help pages.  [OK]Here's how I get the Docker IP:docker inspect --format '{{ .NetworkSettings.IPAddress }}' $(docker ps -q)172.17.0.2Then I enter on the Brick:  172.17.0.2:1999which waits for about 15 seconds before issuing the error cited above.Any help?

WizGeek, 2017-04-06:
Additionally, here's the Docker container list I'm working with:$ docker ps -aCONTAINER ID        IMAGE               COMMAND                  CREATED              STATUS                     PORTS                    NAMES10560a5f66fd        shms/robertalab     "/robertalab-maste..."   About a minute ago   Up About a minute          0.0.0.0:1999->1999/tcp   practical_bhabha33a749fb2cee        shms/robertalab     "/robertalab-maste..."   7 days ago           Exited (137) 7 days ago                             admiring_easley46fe572724c7        shms/robertalab     "/robertalab-maste..."   2 weeks ago          Exited (137) 2 weeks ago                            romantic_hawking499227053eb2        shms/robertalab     "/robertalab-maste..."   2 weeks ago          Exited (137) 2 weeks ago                            eager_volhard2fb87b5d8b3b        shms/robertalab     "bash"                   2 weeks ago          Exited (0) 2 weeks ago                              gifted_mclean

bjost2s, 2017-04-06:
Thanks.
From what you have reported, you tried the ip from the docker server. For me the following worked:
get the ip from your windows laptop from the local network, wifi or lan. I checked it on my windows machine in the terminal with `ipconfig`
EV3: connect to the <ip from windows machine>:1999. If the robot gets an answer from the server, it will show the token for ca. 5 min.
open a browser on the windows machine and open localhost:1999
There I could connect to the robot via the token.
the command 
docker run -p 1999:1999 shms/robertalab
is forewording the server from the docker machine to the laptop. So maybe 172.17.0.2 is the internal ip from the docker server, but you need to connect to the forwarded ip from your home network. Most likely something with 192.....

WizGeek, 2017-04-06:
Still no success.Here's what I tried on the Brick:Laptop's "Ethernet adapter Ethernet" IP:  On Brick:  192.168.1.28:1999 => timeout error message (see below)Laptop's "Ethernet adapter vEthernet (DockerNAT)" IP:  On Brick:  10.0.75.1:1999 => timeout error message (see below)In both cases upon entering the IP:Port, the Brick timed out (~15 sec.) with:|  Open Roberta Lab|  is unreachable!|  |  Reason:|  No access from|  this network!|   (Press any key)In both cases upon entering the token (before Brick timed out), the Chrome Browser OR gave an immediate error:|  Attention||  Check if your robot is switched on and|  connected to the server.|  If you have problems to do this, please|  have a look at our help pages.|  [OK]-------- Original Message --------
Subject: Re: [OpenRoberta/robertalab] Docker Container Image
Configuration for Windows 10 (#448)

bjost2s, 2017-04-07:
Thanks. From what you have reported, you tried the ip from the docker server. For me the following worked: get the ip from your windows laptop from the local network, wifi or lan. I checked it on my windows machine in the terminal with ipconfig EV3: connect to the :1999. If the robot gets an answer from the server, it will show the token for ca. 5 min. open a browser on the windows machine and open localhost:1999 There I could connect to the robot via the token. the command docker run -p 1999:1999 shms/robertalab is forewording the server from the docker machine to the laptop. So maybe 172.17.0.2 is the internal ip from the docker server, but you need to connect to the forwarded ip from your home network. Most likely something with 192.....

WizGeek, 2017-04-07:
Please see my previous comment--your latest comment is a duplicate of your earlier comment. In my most recent previous comment, you'll see I've tried both the laptop IP and the virtual Docket NAT IP without success.  Shall I clarify further?

WizGeek, 2017-04-12:
A few things:

1. See latest post about latest attempts.
2. What is your response to "docker port <container-id>" ?
3. Is your Docker server using Hyper-V or a different virtualizer?

WizGeek, 2017-04-13:
Here's ALL the details of my environment.  Your help truly is appreciated by the student and me.
```
---------------------------------------------------------
Laptop Environment
 * Laptop:  ASUS G750JX-TB71, 16GB RAM, i7-4700HQ
 * Windows 10 Pro 64-bit (Version 10.0.14393 Build 14393)
 * Hyper-V (Version 10.0.14393.953)
 * Docker (Version 17.03.1-ce, build c6d412e)

---------------------------------------------------------
Topology:

  {Internet}=={modem}=={wireless-router}
                             :
     ........................:
     :
  {laptop-wireless}=={Hyper-V}=={Docker-container}=={Linux}=={RobertaLab}
 
---------------------------------------------------------
Hyper-V:
 * Default settings
 * MobyLinuxVM | Running | 2048 MB
  (cfg ver. 8, gen 2)  

---------------------------------------------------------
ipconfig:
 * Ethernet adapter vEthernet (DockerNAT):
     Connection-specific DNS Suffix  . :
     IPv4 Address. . . . . . . . . . . : 10.0.75.1
     Subnet Mask . . . . . . . . . . . : 255.255.255.0
     Default Gateway . . . . . . . . . :

 * Wireless LAN adapter Wi-Fi:
     Connection-specific DNS Suffix  . : fios-router.home
     Link-local IPv6 Address . . . . . : fe80::8d9:5450:c084:fef2%9
     IPv4 Address. . . . . . . . . . . : 192.168.1.175
     Subnet Mask . . . . . . . . . . . : 255.255.255.0
     Default Gateway . . . . . . . . . : 192.168.1.1

---------------------------------------------------------
EV3:
 * Wireless IP:   192.168.1.196
 * Wireless USB:  Edimax EW-7811Un
 * Open Roberta:  2.2.0
 * EV3 Runtime:   2.2.0
 * leJOS version: 0.9.0-beta
 * 2017-02-08 07:03:498
 
---------------------------------------------------------
Dockerfile:
##===============##================================================
##  OpenRoberta  ##
##---------------##
## To build>        docker build -t shms/robertalab .
## To probe/debug>  docker run -it shms/robertalab bash
## To run>          docker run -p 1999:1999 shms/robertalab
##------------------------------------------------------------------
FROM ubuntu

RUN apt-get clean -y && apt-get -y update && apt-get -y upgrade 
RUN apt-get install -y git phantomjs maven
RUN apt-get clean -y

RUN apt-get install -y wget unzip && \
    wget https://github.com/OpenRoberta/robertalab/archive/master.zip && \
	unzip master.zip && \
	rm master.zip

RUN apt-get install -y default-jdk iputils-ping

[... 146 lines cut ...]
  |            "GlobalIPv6PrefixLen": 0,
  |            "IPAddress": "172.17.0.2",
  |            "IPPrefixLen": 16,
  |            "IPv6Gateway": "",
  |            "MacAddress": "02:42:ac:11:00:02",
  |            "Networks": {
  |                "bridge": {
  |                    "IPAMConfig": null,
  |                    "Links": null,
  |                    "Aliases": null,
  |                    "NetworkID": "3ec5038f109efb893d8dcb3cc89a20e98857a0dc930af063763b74f25f1907f3",
  |                    "EndpointID": "21db51a0ce38e90c0ee6d6e329f4fba1335d38146e1ca8e297a08c4b8642a636",
  |                    "Gateway": "172.17.0.1",
  |                    "IPAddress": "172.17.0.2",
  |                    "IPPrefixLen": 16,
  |                    "IPv6Gateway": "",
  |                    "GlobalIPv6Address": "",
  |                    "GlobalIPv6PrefixLen": 0,
  |                    "MacAddress": "02:42:ac:11:00:02"
  |                }
  |            }
  |        }
  |    }
  |]
```

WizGeek, 2017-04-14:
Based on the environment and configuration information above, I've tried all of the following IP/Port values on the EV3 without success; they all timeout on the EV3 in 10-15 seconds of the token appearing:
```
 * DockerNAT: 10.0.75.1     and 10.0.75.1:1999
 * Wireless:  192.168.1.175 and 192.168.1.175:1999
 * Container: 0.0.0.0       and 0.0.0.0:1999
 * Image:     172.17.0.2    and 172.17.0.2:1999
```
It's not clear in the documentation if the port suffix needs to be entered or not.

WizGeek, 2017-05-15:
Any help on this?  Simply put, I need to know how to determine what the correct IP address is I should enter into the EV3.  Is it the Docket NAT, laptop wireless, Docker container, etc?

(Please refer to two previous responses for more details to recreate the condition.)

VinArt, 2017-05-15:
Hi, I have not tried (yet) running docker on windows, but usually if you need to use a service inside of the container you need to use it's ip, for example log in to container and run ifconfig, or inspect as I see above. Maybe you could try ssh-ing into container to see if it is accessible by shell.. Also you can forward a port with ssh with -L option. Also: will it help to play with network settings like bridged/NATed connection?

waja, 2017-08-10:
You should have a look into https://hub.docker.com/r/exmatrikulator/robertalab/~/dockerfile/

bjost2s, 2017-11-25:
please discuss further here #699

closed 2017-11-25 by bjost2s