OpenRoberta/robertalab #449: Edge Save Target As not working

cvejoski, 2017-03-20:
For all the embed systems downloading the hex code using _save target as_ is not working. The save dialog is not opened.

bjost2s, 2017-03-21:
seems to be an issue of edge
https://developer.microsoft.com/en-us/microsoft-edge/platform/issues/4282810/
someone pointed out, that there is a limitation of 2000 chars limitation

bjost2s, 2017-03-21:
Think we cannot limit the hex file

cvejoski, 2017-03-21:
No, we cannot :). Maybe we should have alternative way of downloading the hex on Edge

bjost2s, 2017-03-21:
If you download it directly, (accepted once the hint that you have changed your download folder) it will work
[image]
so I will check for edge and treeted it as accepted download folder

bjost2s, 2017-03-21:
And I will take this into account
https://developer.microsoft.com/en-us/microsoft-edge/platform/issues/7260192/
:-(

bjost2s, 2017-03-21:
fix is only a workaround, we should think about putting the browser and/or device system into the GUI state to react in general more specificly

closed 2017-03-23 by bjost2s