OpenRoberta/robertalab #451: microbit: "get value ambient light" has no code representation

larspelz, 2017-03-22:
Hey @all!

When I use the block "get value ambient light" in the microbit version of NEPO, no code will be generated for the block. The program can be downloaded but yields a syntax error message on the microbit.

Greetings - Lars Pelz

cvejoski, 2017-03-22:
micropython microbit does not provide "ambient light", therefore there is no implementation in NEPO.

Probably we have to remove this block from the toolbox.

bjost2s, 2017-03-22:
There is already a ticket for this here https://github.com/bbcmicrobit/micropython/issues/248, but I don't think that they are working on it.

larspelz, 2017-03-22:
So currently there is no way to retrieve the light sensor's value of the microbit in NEPO?

bjost2s, 2017-03-24:
We will check micropython and add it back, if it it is available

closed 2017-03-24 by bjost2s