OpenRoberta/robertalab #453: EV3 - Gyroscope reset bug

Dittse, 2017-03-23:
As I tried to drive a square with the EV3 I noticed the following issiue:
If the "reset gyroscope" block gets used you can tell by the sensordata that it gets a frequent rest. Even when it's not supposed to yet.
In this programm for example the EV3 just keeps spinning but never reaches a gyroscope angle of 90°.
[SquareBugEV3.zip](https://github.com/OpenRoberta/robertalab/files/864780/SquareBugEV3.zip)

[image]

ensonic, 2017-03-23:
Is this using lejos or ev3dev? For ev3dev I just fixed a bug OpenRoberta/robertalab-ev3dev#38

Dittse, 2017-03-23:
So far this has only been tested with lejos

WizGeek, 2017-03-23:
It may depend upon gyro orientation--right-side up or upside-down.  Going left may produce a negative angle, so your test might have to be "< -90".

Dittse, 2017-03-24:
The gyro returned positive angles for my configuration. It just receives a frequent reset before it can reach 90°. If you delete the block “reset gyroscope” the EV3 drives forward, turns and then just drives forward three more times.

cvejoski, 2017-07-28:
The logging calls the reset gyro every second, since as used sensor is sent the gyro sensor with mode RESET :/.

This is a part of the generated code:

`public class NEPOprog {
    private static Configuration brickConfiguration;

    private Set<UsedSensor> usedSensors = new LinkedHashSet<UsedSensor>(Arrays.asList(new UsedSensor(SensorPort.S2, SensorType.GYRO, GyroSensorMode.ANGLE), new UsedSensor(SensorPort.S2, SensorType.GYRO, GyroSensorMode.RESET)));

    private Hal hal = new Hal(brickConfiguration, usedSensors);

    public static void main(String[] args) {
        try {
             brickConfiguration = new EV3Configuration.Builder()
                .setWheelDiameter(5.6)
                .setTrackWidth(18.0)
                .addActor(ActorPort.B, new Actor(ActorType.LARGE, true, DriveDirection.FOREWARD, MotorSide.RIGHT))
                .addActor(ActorPort.C, new Actor(ActorType.LARGE, true, DriveDirection.FOREWARD, MotorSide.LEFT))
                .addSensor(SensorPort.S1, new Sensor(SensorType.TOUCH))
                .addSensor(SensorPort.S2, new Sensor(SensorType.GYRO))
                .addSensor(SensorPort.S3, new Sensor(SensorType.COLOR))
                .addSensor(SensorPort.S4, new Sensor(SensorType.ULTRASONIC))
                .build();
            new NEPOprog().run();
        } catch ( Exception e ) {
            Hal.displayExceptionWaitForKeyPress(e);
        }
    }`

closed 2017-09-08 by bjost2s