OpenRoberta/robertalab #457: Bug in ora.sh  --export

corvus-albus, 2017-03-25:
Hi,
I love this project, keep on going!

There is a tiny little bug in the script ora.sh

line 138 
actual:
       cp -r --parents resources "${exportpath}/$Robot/"
has to be changed in
      cp -r --parents resources "../${exportpath}/$Robot/"

The working directory is src-root/$Robot when cp is executed but $exportpath starts from src-root.

Cheers corvus-albus

bjost2s, 2017-03-25:
Thank you! Do you mean branch master or develop?

corvus-albus, 2017-03-25:
I installed the master branch.

rbudde, 2017-03-28:
Hi,
thank you for your nice comments and your bug report.

In the ora script I made the (naive) assumption, that the export directory is given as an absolute path. In that case everything is ok. I assume, that you have used a relative path (which is fine, of course) and then the cp command is broken.

I used the idiom
ABS_PATH=`cd "$EXPORT_PATH";pwd`
to get an absolute path and used that in the script. Now everything works fine.

The change will be released (in master and develop) soon (probably this week).

Cheers Reinhard

rbudde, 2017-03-28:
Note: the backtick in the shell command has been interpreted by markdown as inline quote. I always get caught in that trap :-)

closed 2017-04-06 by bjost2s