OpenRoberta/robertalab #458: support for third party EV3 sensors

bjost2s, 2017-03-25:
- [ ]  NXT IRLink Sensor, [info](http://www.hitechnic.com/cgi-bin/commerce.cgi?preadd=action&key=NIL1046)
 - [ ]  HiTechnic compass sensor, [info](http://www.hitechnic.com/cgi-bin/commerce.cgi?preadd=action&key=NMC1034)
- [ ]  Dexter compass sensor, [info](https://www.dexterindustries.com/manual/dcompass/)
- [ ]  Mindsensors compass sensor, [info](http://www.mindsensors.com/ev3-and-nxt/37-compass-for-nxt-or-ev3)
 - [ ]  HiTechnic IR-Seeker V.2, [info](https://www.hitechnic.com/cgi-bin/commerce.cgi?preadd=action&key=NSK1042)
 - [ ]  Adapter für Grove-Sensoren für EV3 oder NXT [info](https://www.generationrobots.com/de/402125-adapter-fur-grove-sensoren-ur-ev3-oder-nxt.html)
 - [ ]  Mindsensors Light Sensor Array for NXT or EV3,[info](http://www.mindsensors.com/ev3-and-nxt/47-light-sensor-array-for-nxt-or-ev3)
 - [ ]  Dexter Thermal Infrared (TIR), [info] (https://www.dexterindustries.com/manual/thermal-infrared-sensor/)
 - [ ]  HiTechnic NXT EOPD, [info](https://www.hitechnic.com/cgi-bin/commerce.cgi?preadd=action&key=NEO1048)

ensonic, 2017-04-24:
Here is the list of supported sensors for ev3dev:
http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensors.html
All the HiTechnic sensors are on the list. Also the 'Mindsensors Light Sensor Array for NXT or EV3' is there. Also the Grove-Adapter won't need a driver for the adapter, but for whatever is plugged.

The others would need testing and reporting to this list.

bjost2s, 2017-04-24:
The list was ordered by user priority.

ensonic, 2017-04-24:
Sorry changed the order back and also mention that fact. How did you determine the order? This would have been useful to mention.
I'd would suggest to remove the "Adapter für Grove-Sensoren", this does not need driver support and cannot be supported generically.

bjost2s, 2017-11-22:
@ensonic could we please check which sensors could be implemented in all systems? lejosv0, lejosv1 and ev3dev?
It makes us a bit easier to choose one out of them

ensonic, 2017-11-24:
Do we have a list which ones are supported by lejos[0,1}? Should I try to reformat the top coment into a table with extra columns for the ev3 os variants?

boonto, 2018-01-29:
I updated the top comment with the support table. The "untested" ones for lejos say that in the documentation, but I guess they should work most of the time. I could not find anything regarding the Grove sensors for lejos.