OpenRoberta/robertalab #465: Usability OR Lab on iPAD

Thorlogy, 2017-04-03:
As a user i would like to have the same usability on my iPAD like on my PC. But some buttons needs two clicks - some only one at the iPAD.

OS version and different browsers have no influence.

bjost2s, 2017-04-05:
Could you please provide an example for each of this buttons types, one and two clicks?

MaikBaeumer, 2017-06-21:
All the buttons on the top (edit, robots, help, user, language) and all the buttons in the simulator view (start, change the scene, open/close the robot's view, open/close the sensors's data view, upload your own ...) need two taps to activate on the iPad. 
First tap opens the tooltip and second tap activates the button. The tooltip does not disappear and hides other buttons, which can be taped anyway.

closed 2018-07-17 by bjost2s