OpenRoberta/robertalab #467: Variety of list category reduced

Dittse, 2017-04-03:
The Variety of the list category was very limited. Then it was extended and almost had as much list options as EV3. They new options worked more or less (at least “in list … get random” worked). Sadly, they all were cut again and it got more difficult to write complex programs.
I’m hoping to use them again soon to write cool and interesting programs with NEPO.

ensonic, 2017-04-05:
The number of items varies whether you use normal or expert mode. In the new version, there are now two tabs above the category list. Is that solving it for you.

bjost2s, 2017-04-05:
We reduced the Lists blocks functionality for some Calliope blocks, because some couldn't be supported. We will check again, if we at least add "the get random" again.

cvejoski, 2017-07-21:
Some math operations on list are now implemented. See #571

closed 2017-07-21 by cvejoski