OpenRoberta/robertalab #468: Can't find documentation on how to add custom robot

sommmen, 2017-04-04:
Hi,

My company wants to add their own robot to OR, and i'm tasked with writing down some info on how to implement this. Going through the docs I cannot find a lot about how to implement a custom (arduino) robot. We want to run the webserver on the robot itself, so not use the platform. Could you provide me with some information where to start, or update the wiki with some directions?

Thanks in advance.
(btw, great job  both back and frontend are astethically pleasing. So well organized 👍 )

tl;dr:
How to add a custom robot to the OR platform.
- hosting OR on robot itself

ensonic, 2017-04-05:
Awesome, if the robot is already on the marked, it would be awesome to know more details.

As of now,w e don't have such a doc yet. We're still in the process of tweaking the robot interfaces. All current robot plugins sit in the code base for easier refactoring.

Regarding to your plan to run the webserver on the robot itself, the code is written in java and too heavy to run on an arduino. It works on a raspberry pi though.

sommmen, 2017-04-05:
Thanks for your reply.

Could you perhaps direct me to where i would need to change some things in order to add a robot?

bjost2s, 2017-04-05:
Hi, we are currently integrating arduino into the Open Roberta lab. Please have a look into the RobotArdu project which will be very simular to all other ardunio robot projects. Main changes  have to be done in robertalab/RobotArdu/src/main/java/de/fhg/iais/roberta/syntax/codegen/Ast2ArduVisitor.java as you may be would like to add your own arduino libraries.
If you would like to add a robot to the Open Roberta Lab, please make sure you present the robot here before you start or write a detailed proposal, so that we can discuss together, how NEPO can look like for your special robot. 
 If you would like to run our server on the robot itself, you can fork the repo and customize it to your needs. 
Please introduce your robot system, maybe just a link, we are curious!

bjost2s, 2017-04-07:
Hi again, any infos or questions from your side?

sommmen, 2017-04-07:
We're still in the development phase, the original robot we have now will be significantly altered to incorporate OR or we'll start fresh. So i got nothing to show yet ;)

I'm just laying some ground work for now, what's needed to realise this, what hardware is required to run the server etc.

Out of curiosity i thought the platform was an attempt at creating a place where everyone could add a robot, a bit like Scratch but for educational robots. Is this a goal in the future? And what is the direction of OR?

sommmen, 2017-04-09:
Also i seem to be getting this error (./ora.sh --start-from-git):
`2017-04-09 20:54:27,868 [main] session-id=[] user-id=[] robot-name=[] ERROR d.f.iais.roberta.main.ServerStarter - Abort: no usable version for upgrade to 2.2.2 from one of [2.2.0, 2.1.0, 2.2.1]
`

Running on win10. Any ideas?

cvejoski, 2017-04-09:
In OpenRobertaServer copy the dbBase folder to db-2.2.2. This should do the job

sommmen, 2017-04-09:
@cvejoski that worked, thanks!

bjost2s, 2017-04-10:
"Out of curiosity i thought the platform was an attempt at creating a place where everyone could add a robot, a bit like Scratch but for educational robots. Is this a goal in the future? And what is the direction of OR?"
What Open Roberta is: A platform to offer the graphical programming language NEPO to all interested people, exspecially to kids and newbies. The hurdles to start programming robots should be as low as possible. The whole system is open source, so people are welcome to extend Open Roberta, provide new ideas, new robots .... We understand this repository as a platform, where everybody can contribute and be part of the commiter team. We would like to provide our users a long term support, so a deep commucation and cooperation with the developers and creators/companies of the robots systems is essential. We as Fraunhofer IAIS are the only ones who are allowed to set Open Roberta online the way you see it at lab.open-roberta.org.

What Open Roberta is not: This is not a service. So no matter what you will contribute, we are checking it carefully, before we set it online. Also we asked for a proposal and/or frequend participating in our developer group, if someone is interested in. Please see https://github.com/OpenRoberta/robertalab/wiki/Community.

For full support of a robot system it is obvious, that this system is (maybe partually) open source. 
You can always use this repo as a base of your own developement. Make shure you meet our notices to the licences here https://github.com/OpenRoberta/robertalab/blob/master/Resources/NOTICE in that case.

bjost2s, 2017-04-13:
@sommmen , do you have further questions? Or do you want to give us some more information of what your are planning with the new system?
Otherwise please close this ticket.

sommmen, 2017-04-14:
I willl be working on this project in a few months time, until that i'll close this ticket. 
Thanks for your excellent support :)

closed 2017-04-14 by sommmen