OpenRoberta/robertalab #471: OR Simulator - unexpected behavior

Thorlogy, 2017-04-06:
[image]

Expection: Robot drives backward 20cm then start steering again. 
Result: Robot stops shortly then start steering

bjost2s, 2017-04-06:
The problem of this program ist that you either ask the touch sensor after driving backwards 30cm or after steering in a curve for 20cm . At this point the robot may be just be without contact to the touch sensor because it is rejected from the obstacle. Please try this more reactive program instead:
[image]

cvejoski, 2017-04-07:
It is better to use the ultrasonic sensor instead of the touch sensor, then the behavior of the program will be more obvious.

closed 2017-04-07 by Thorlogy