OpenRoberta/robertalab #486: Cannot work with NAO

nicolaipoehner, 2017-04-24:
Dear all, 

i just found out about the support of Open Roberta Lab with NAO and immediately got started. I was able to install and connect to NAO but when I run a programm NAO doesn't do anything even though I am connected to the robot (at least that's what the green light is supposed to indicate) Any help? Thanks in advance! 
Nicolai

JanisM91, 2017-04-24:
Hi Nicolai,

i will try to help with your issue. Can you provide some more information first?
Which version of the NAO (Hardware + Software) are you using?
Can you send a screenshot of the program you tested?

Additionally please try this simple program on your robot:
[image]

Regards
Janis

nicolaipoehner, 2017-04-25:
Hi Janis,

I'm using a Nao version V5 with Naoqi 2.1.4.13 running on it. The programm I tested was a simple hello world programm looking like this: 

[image]

I also tried your idea alredy but it didn't help.

JanisM91, 2017-04-25:
Hi Nicolai,

thanks for the information your provided.

It seems like the version of your NAO is fine. There can't be a problem with the generated code or the HAL for the NAO. Otherwise NAO should say Error, if the uploaded program can`t be executed.

Therefore i conclude that there is a problem with the file transfer. The green icon just states that the connection between your PC and the Open Roberta Lab is established. It seems like there is a problem with the connection to your NAO. 
Can you evaluate if the IP, username and password are correct. Just trie to log into the web interface of the robot with the information.

Regards
Janis

nicolaipoehner, 2017-05-19:
Hi Janis,

sorry I haven't answered in a while but I've been busy with other things. Well I tried and retried a couple of times but NAO does still not respond to any commands from OpenRoberta. I've double checked the ip address, username and password and the login to NAO's webpage always worked. Any other suggestions what I could try?

Kind regards,,
Nicolai

JanisM91, 2017-05-29:
Hi Nicolai,

meanwhile i checked the connection and communication with the Open Roberta Lab with our NAO and it works just fine. So it doesn`t seem like there is a general problem.
I still think that there is a problem with the communication between your PC and your NAO.

Do you connect the NAO via Ethernet or WiFi? Are there any routers with firewalls in between or is the communication in the network restricted?

I´m looking forward to solve this issue as i haven`t encountered a similar issue, yet.

Regards
Janis

JanisM91, 2017-07-03:
Hi Nicolai,

we made some changes on the backend of the connection program for NAO.
These changes should have fixed your issue. (There was a problem with the transfer of the HAL)
Please download the new version of the software from our wiki and test it with your robot.

I´m looking forward to hear from you.

Regards
Janis

nicolaipoehner, 2017-07-18:
Dear Janis,

it still doesn't work. I has to be a transfer error since I downloaded the Python file and manually uploaded it onto the robot with Filezilla and it worked. However I had to get rid off some errors in a hal.py file which has been on the robot before (missing intended blocks, syntax errors, etc.) This file looked like a collections of definitions to convert the NEPO code to Python code. 
Since we are speaking of connections all the time, I was wondering whether you know how to connect NAO to a WPA2 Enterprise network (if possible?)? I was trying to configure wpa_supplicant.conf file on the robot but don't get it to work. After a little research others seemed to have this problem too. On the old NAO web page you see that NAO detects those networks but I can't connect to them since I can't configure it (type in identity and password, etc.). 

I'm looking forwad to hearing from you.

Regards,
Nicolai

nicolaipoehner, 2017-07-18:
Dear Janis, 
just to anwser your other question: My NAO is connected via Ethernet (plugged in a switch in my office). The network is the university network eduroam. The potential wifis I could use are listed here:
https://www.rz.uni-wuerzburg.de/dienste/kommunikation/wireless_lan/verfuegbare_wlan_netze/

Kind regards,
Nicolai

JanisM91, 2017-08-01:
Dear Nicolai,

thank you for your response. Now the installer in the wiki is up to date and hopefully working for you.(?) In a next step we will not make the hal.py file a ressource but a file that is send from the server. Therefore we don't have to generate a new installer for every change in the hal.py file and the users can use the programm they once installed.
You are quite right with your assumptions. The hal.py file is a collection of definitions and functions that is needed to execute the code generated from the Lab.

Ethernet is fine.

Regarding to http://doc.aldebaran.com/2-1/news/known_issues.html WPA 2 Enterprise is not supported in the current version of NaoQi. But it is listed in the Known Issues so i hope that Softbank Robotics are working on this issue and release a fix. As most universities and companies use the WPA2 Enterprise standard i think it would be nice to have that for NAO.

Kind regards
Janis

nicolaipoehner, 2017-08-01:
Dear Janis, 

unfortunately it is not working since when I'm trying to start my program I get the error message that something is wrong with the download of my program and that I should try again. Any ideas? Sorry for bothering you.

Kind regards,
NIcolai

JanisM91, 2017-08-01:
Dear Nicolai,

there is no need to feel sorry. ;)

Are you using the newest version? If you click on Info -> about it should say version 1.0.0.-SNAPSHOT.
The latest release is here: https://github.com/OpenRoberta/robertalab-naoprogram/releases

You can also try to have a look at the log file. Maybe this helps us to get closer to the source of this issue.
It should be located here: ...\Users\*USERNAME*\AppData\Roaming\OpenRobertaNAO

Feel free to send the log to me, if you want me to have a look please send it to the mail adress in my profile.

Regards
Janis

ProCycon, 2017-09-26:
Hello Janis.
I have to jump in your conversation, because I have the same problem with my NAO und the connection between Open Roberta Lab. 
I am using the latest version of Open Roberta NAO with the Open Roberta Lab online under lab.open-roberta.org.
[image]

My NAO is connected to my machine with ethernet and it seems, like the connection is fine. The green symbol in the application is showing. I checked the login several times and they are correct, because I can access with them the NAO web access. 
[image]

When I try to upload the following programm to my NAO, I get the following message, stating, that the transfer did not work and I have to try again. 

[image]

[image]

Do you have any suggestions, where I can look to solve my problem? If you need more informations, I will provide them.

With best regards,
Daniel

bjost2s, 2017-09-29:
There was a problem in the NAO connection program. It is fixed now. Please checkout the newest installer here https://github.com/OpenRoberta/robertalab-naoprogram/releases and give it a try.

fixed with https://github.com/OpenRoberta/robertalab-naoprogram/commit/de0fcbb75b7ce0656b5f72b2532311adecba8ee1

closed 2017-10-02 by cvejoski

ProCycon, 2017-10-12:
Thanks bjost2s.
The new version works!