OpenRoberta/robertalab #487: Calliope 2016/2017: Microphone block yields value ~520

larspelz, 2017-05-02:
Hi openroberta team!

I tried the microphone block (see screenshot below) but I'm only getting a constant value around 520. I tried both Calliope versions, which behaved the same. Thanks in advance for looking into this problem!

[image]

cvejoski, 2017-05-11:
This is a bug in Calliope firmware as soon as they fix it we will update the firmware

bjost2s, 2017-05-14:
see https://calliope-mini.myjetbrains.com/youtrack/issue/MINI-51

bjost2s, 2017-07-03:
in version 2.2.7 we map the output from the calliope to 0-100, so old  hardware version will get a value between 0-10 and newer versions 0-100 (not testet, no device available :-() to simply messure the volume.

larspelz, 2017-07-04:
I tested it and can confirm it's working. Thanks for fixing!

closed 2017-07-21 by cvejoski