OpenRoberta/robertalab #492: Arduino create agent integration

elhele, 2017-05-09:
As a user I'd like to upload programs without using USB program and manual input of the token.

For this an Arduino create agent can be used. It is easier to install, launch and it doesn't require a token. 

User would have to download and install the Arduino Web Editor plugin (http://create.arduino.cc/getting-started/plugin). The plugin would be running in a tray (example for linux):
[image]

Then user should go to Open-Roberta lab, choose a robot icon and press "connect":
[image]

After this user'll see the following pop-up:
[image]
where would be listed all Arduino devices with ports attached to the user's computer. After choosing the device user can run a program on the robot. 

Proposed solution:

- []  Provide Arduino create agent preconfigured with Open-Roberta public key
- []  Provide install script for *nix-based OSs
- []  Implement web-sockets for listing available Arduinos with ports
- []  Implement GUI integration

VinArt, 2017-07-31:
There is a problem with SSL certificates: they must be present on the system (so we need to use wss instead of ws connection) AND the user must accept this local self-signed certificate.

VinArt, 2017-08-15:
Testes today a fresh install of create agent on both windows and linux. For some reason it is not shipped with avrdude and thus fails to flash boards. The problem is in /home/<username>/.arduino-create folder. The file "installed.json" should be modified to look the following way:

`{"apilevel":"v1","avrdude":"/home/<username>/.arduino-create/arduino/avrdude/6.3.0-arduino9","avrdude-6.3.0-arduino9":"/home/<username>/.arduino-create/arduino/avrdude/6.3.0-arduino9"}`

On Linux platform. On Windows it should be edited in the similar way, namely:

`{"apilevel":"v1","avrdude":"C:/Users/<username>/.arduino-create/arduino/avrdude/6.3.0-arduino9","avrdude-6.3.0-arduino9":"C:/Users/<username>/.arduino-create/arduino/avrdude/6.3.0-arduino9","windows-drivers":"C:/Users/<username>/.arduino-create/arduino/windows-drivers/1.8.0","windows-drivers-1.8.0":"C:/Users/<username>/.arduino-create/arduino/windows-drivers/1.8.0"}`

This implies that the user is able to copy the appropriate version of avrdude and for Windows the drivers to the aforementioned locations.
For reasons unknown, the current installer does not do this automatically. If we will be providing our own version (and we will, hence the configuration is different) then we can as well copy the needed resources and configuration.

VinArt, 2017-08-15:
UPD: it is possible to send a command over websocket to create-agent to download the needed tool, like avrdude, and that will automatically change the json file. But this requires Internet connection.

closed 2017-09-25 by cvejoski