OpenRoberta/robertalab #495: Converter Calliope pxt program to NEPO

Thorlogy, 2017-05-10:
As a teacher it would be nice to be able to convert Calliope programs which are prgrammed with the pxt editor into NEPO programs.

larspelz, 2017-05-13:
As I see it, this will be impossible. NEPO has features (like lists) that PXT doesn't have, whereas PXT also has features that don't exist in NEPO (event handling, e.g. "when button pressed"). As a teacher, I also have to decide which platform to use in my lessons and then stick to it.
As it is, we still have to wait for a graphical programming language that has everything built-in and supports each and every platform, too...

bjost2s, 2017-05-14:
What about all the other graphical editors? 

- ev3 lego
- scratch
- ...

and what about the other way round?

ensonic, 2017-05-17:
Like what Lars said. I don't think it is feasible to convert such projects. It might work for trivial projects, but for those one can also just recreate them. What we can do is to have docs to address how to redo a sketch from a different tool.

bjost2s, 2017-07-14:
We try to explain in our docs how nepo is working as an imperative programming language. If people want to redo something from another programming language, they can easily start from the problem/task itself. 
We can describe what our blocks are doing, which values or actions they deliver. We already do it in our help and it is worth to have a look there and improve this. I don't think we should explain it the other way round. Also the list of the different tools is maybe infinitely long.

bjost2s, 2017-10-09:
@Thorlogy please close this or, if you still would like to have this feature I can put it into the wish list. Thank you

closed 2017-10-10 by Thorlogy