OpenRoberta/robertalab #515: NXT: problem with import

Mathias573, 2017-06-06:
- I made a program to move forward:
[image]
- transfered the program to robot and start: robot moves forward (ok)
- saved the program under "my programs" and exported it to local drive ("test_forward")
- made a new program to turn around:
[image]
- transfered the program to robot and start: robot turn around (ok)
- saved the program under "my programs" and exported it to local drive ("test_turn")

- now import the first program ("test_forward") and transfer it to robot and start: robot turn around (!)
- open the first program from "my programs", transfer to robot and start: robot turn around (!)
- edit a new program in the editor like the first program, transfer to robot and start: robot moves forward (ok)

In simulation there is no problem with import-function. Robot-simulation acts as expected. Only if I transfer an imported program or saved program direct to the robot it seems the robot don't recive the new one. If I write the program new in the editor everything is ok.

cvejoski, 2017-06-07:
When you download a program on the NXT you have to choose the right name of the program. 
It is possible that you are in the menu state to run the program "test_turn" and when you download the new program "test_forward" you run the old program. You have to choose the name of the new program.

I've tested the workflow that you describe here and everything works fine.

Mathias573, 2017-06-07:
hi cvejoski,

you are right: on my nxt I found 3 programs: "NEPOprog", "test_forward" and "test_turn". Everything works fine. So if I transfer a program with a name it's stored on nxt under his name and a new one without a name is transfered as NEPOprog. Somethimes it's so easy. Thank you for test and respond.

closed 2017-06-07 by Mathias573