OpenRoberta/robertalab #517: NXT: no warning message for long filenames

Mathias573, 2017-06-07:
If I load a file with a filename with max. 15 characters and transfer the program to robot NXT there is no problem.
If I load a file with a filename with exact 16 characters and transfer the program to robot NXT I get an error-message:
[image]
If I load a file with a filename with 17 or more characters and transfer the program to robot NXT I get no error-message and everything seems ok (but on NXT is no new file because of the long filename).

Btw: I can save the programs in the cloud under "my programs" with more then 15 characters without an warning-message. If somebody only play with simulation there is no problem. But when he transfer the program to NXT he gets an error-message (if you fix it). Perhaps he is than confused? Can you implement a message before downloading to NXT with the info, that long filenames don't work?

bjost2s, 2017-06-07:
TODO fix inconsistency of server and clientside tests for input! We should remove the test of the number of characters for program names or at least allow longer names to allow  descriptive names.

cvejoski, 2017-07-17:
fixed with commit 9e2c18a6406be8ddb74952aa88443466924c4952

closed 2017-07-17 by cvejoski