OpenRoberta/robertalab #523: Bob3 integration

VinArt, 2017-06-12:
Bob3 is a atmel-based board, with the following components:
* 2 white LEDs
* 2 RGB LEDs
* IR sender/receiver
* temperature sensor
* 2 modular touch sensors (hands)
* 5 solderable jumpers (set binary code 0-31)

For the integration following functions are needed:
- [ ] setEye(color)
- [ ] setLED(on/off)
- [ ] sendString()
- [ ] receiveString()
- [ ] getTemperature()
- [ ] getArm(left/right, up/middle/bottom)
- [ ] getCode()

Bob3 can be flashed with arduino create agent. It is mandatory to erase the chip during each flash with -e option of avrdude.

elhele, 2017-06-12:
There is already an issue #345 for this

VinArt, 2017-06-12:
#345 I guess I was not in this project at that point =)
Anyway, there exists a thing, that it is not possible, correct me if I am wrong, to re-use blocks from other projects, and for now, I need to use a block from mbed sensors in ardu project and since a block cannot be mapped to two classes, I would have to clone the block, which is not nice... Move Bob3 to mbed project? But it is arduino-compatible. Frustration, that is what it is..

bjost2s, 2017-06-13:
please use #345

closed 2017-06-13 by bjost2s