OpenRoberta/robertalab #530: Calliope and Microbit Simulation: Display error

Schlingloff, 2017-06-17:
Dear Developers,
Here is a bug in the Calliope and Microbit Simulation.
The attached program should have two LEDs (say, P1=(x1,y1) and P2=(x2,y2)) blinking at different frequencies.

[SimulationDisplayError_Calliope&Microbit.xml.txt](https://github.com/OpenRoberta/robertalab/files/1082559/SimulationDisplayError_Calliope.Microbit.xml.txt)

It works perfectly correct when downloaded to the device.
However, in the simulation, depending on the setting of the points, strange things happen:
Whenever the two LEDs are in the same row or column, only one flashes. 
E.g., for points P1 = (0,0) and P2=(1,1), everything works fine, both LEDs blinking.
For P1=(0,0) and P2=(0,1), P1 blinks and P2 stays lit.
For P1=(1,0) and P2=(0,1), P1 stays off and P2 stays lit.
The problem exists with both versions of the Calliope, and with the Microbit Simulation.
Kind regards,
Holger Schlingloff

VinArt, 2017-07-13:
I found an interesting thing. With your program only diagonally placed LEDs blink (e.g. 0,0 1,1 2,2)

VinArt, 2017-11-27:
Oh, this might be realted to thi bug here: #713 I will see if that fixed this too.

closed 2018-01-30 by bjost2s