OpenRoberta/robertalab #532: Update welcome tour

elhele, 2017-06-19:
There were many improvements in enjoyhint library. We need to adjust our welcome tour to those changes.

bjost2s, 2017-07-27:
The new version of enjoyhint doesn't recognize the menu as part of the  website. Also the escape button has disappeared.
[image]

elhele, 2017-07-31:
@bjost2s  It was just a wrong z-index. Fixed it with 41eb0f8

elhele, 2017-08-11:
Now tour seems to be working fine except for the devices with touch screen:
when the tour is supposed to be finished when "sim" button is pressed:
[image]
instead of going back to the list of robots it just closes the simulation and stays in a welcome tour:
[image]

closed 2017-10-02 by bjost2s