OpenRoberta/robertalab #535: fix calliope microphone

bjost2s, 2017-06-20:
map the analog output to 0-100, slightly
see http://www.elektronik-labor.de/Microbit/Calliope2.html#kl 
new calliopes will have an amplification of 10, so this is the majority

larspelz, 2017-07-04:
Happy to see progress on this! I tested it on both the old and new boards and can confirm it's working. However I don't get raw values higher than ~25 even on a new board. I attached my test program.
[nepo-mic-test-c17.zip](https://github.com/OpenRoberta/robertalab/files/1121606/nepo-mic-test-c17.zip)

bjost2s, 2017-07-07:
Normally the microphone needs to have an amplification on the hardware side. Here it is explained in detail: http://www.elektronik-labor.de/Microbit/Calliope2.html
Unfortunately I do not have a calliope where the problem with the resistor is fixed, but I 've heard that newer once will have it. Once I have one, I could will improve the mapping if necessary.

closed 2017-07-07 by bjost2s