OpenRoberta/robertalab #538: Different behavior Sim vs. Real EV3 robot - lejos

Thorlogy, 2017-06-21:
This program causes different behavior 

[image]

In OR Sim it works fine. On the real EV3 robot - using lejos firmware - the robot does not move (or moves very slowly)

cvejoski, 2017-06-21:
The problem is that "drive forward" block is called two often. If you put it outside repeat it will work fine.

Thorlogy, 2017-06-21:
1. yes i know - but the problem is the different behavior of sim vs real robot 
2. putting the block outside repeat loop the behavior is is a different program with a different behavior

bjost2s, 2017-06-21:
[image]
Please program "events" in this way. NEPO is exspecially designed for this. This program will work on both, real and sim, the same. Why?
The drive forwards block needs time to be executed. Go through the blocks from the beginning to the end.
The drive forwards block will only be executed after the robot bumped into an obstacle.
We already have talked about this.

closed 2017-06-21 by bjost2s

Thorlogy, 2017-06-21:
The problem for the user is the different behavior Sim vs Real robot.

ensonic, 2017-06-22:
The simulator is not a physical simulation. I agree with @bjost2s suggestion.It is not feasible to model the same behavior on the simulator. There we'd need to add a micro delay for starting the motors or something like it.