OpenRoberta/robertalab #543: Blocks disappear when uploading to EV3 leJos (or when simulate)

patrickschaerer, 2017-06-25:
When I try to upload a more complex programm to EV3 most of the blocks disappear in the browser window. The program loads to the EV3 perfectly.
To see all the blocks again I need to reload the page, login and load the program.
before uploading:
[image]
after uploading:
[image]

bjost2s, 2017-06-26:
When this had happened, could you please have a look into the help menu - logging? If there is a red marked line, please send us a screenshot. Also can you attach the exported xml from the program? Thank you

patrickschaerer, 2017-06-26:
This is the log (created without connection to EV3 but with simulate) ... it seems there is a bug in my syntax
`NO.
TYPE
MESSAGE
NO.
TYPE
MESSAGE
25	[[TIME]]	10 msec: help clicked
24	[[TIME]]	2 msec: log list activated
23	[[ERR ]]	[[TIME]] 1 msec: run program 'SimonSays2' with configuration 'EV3basis', then EXCEPTION: SyntaxError: Unexpected token var
22	[[ERR ]]	[[TIME]] 62 msec: run program 'SimonSays2' with configuration 'EV3basis', then EXCEPTION: Connection checks failed.
21	[[TIME]]	240 msec: tabProgram clicked
20	[[TIME]]	7 msec: back to previous view
19	[[TIME]]	28 msec: help clicked
18	[[TIME]]	3 msec: log list activated
17	[[TIME]]	230 msec: tabProgram clicked
16	[[TIME]]	9 msec: load program 'SimonSays2' owned by 'Pschaerer'
15	[[TIME]]	6 msec: Load program from listing double clicked
14	[[INFO]]	loadFromList SimonSays2
13	[[TIME]]	20 msec: load program list`

[SimonSays2.xml.txt](https://github.com/OpenRoberta/robertalab/files/1101209/SimonSays2.xml.txt)

bjost2s, 2017-06-26:
thank you, it seems that the create empty list block causes the problem. This little test program is behaving similar:
[image]

patrickschaerer, 2017-06-27:
I changed the code to your example but it did not correct the problem
[image]
[image]
[SimonSays3.xml.txt](https://github.com/OpenRoberta/robertalab/files/1104781/SimonSays3.xml.txt)

elhele, 2017-06-27:
The problem is caused by array rewriting. it wouldn't matter if it is empty or any other array. This would't work either: 
[image]

Also, it is a problem of all robots, not only ev3.

VinArt, 2017-06-30:
Just for the information: the problem was in assignment block, when it was doing transformations, it capitalised the first letter of block name and made everything else lowercase, but the first letter of type (which is part of blockname) should have been capital.

closed 2017-07-07 by bjost2s