OpenRoberta/robertalab #544: Cannot compile program when using existing symbol names for variables

mdondorff, 2017-06-28:
Reproduction:
Go to https://lab.open-roberta.org/#
Choose caliope 
Add a variable by clicking '+' on start and call it 'max'
Hit the 'Run on caliope 2017' / compile button.

Expected:
- Compiles, or
- Error message one can act upon

I think there will be a multiple of symbols which cannot be used as variable names.

This does not reproduce on e.g. 'micro bit' configuration.

cvejoski, 2017-06-29:
The name **max**  is already used for a function name that is already imported. We will put the **max** in the reserved keywords for calliope

mdondorff, 2017-06-29:
Thanks! 

Any chance to know which other names would be good to add to that list as well? (I'm looking at you, 'min'!)

bjost2s, 2017-07-01:
We have to go through all included libraries and check the function names. For math you can have a look here http://www.cplusplus.com/reference/cmath/

larspelz, 2017-07-04:
I ran into the same problem recently. Is there any chance to show a hint in the error message, e.g. "Check your variable names?" Or would it be possible to prefix two underscores to any variable name in the generated code to avoid this problem?

bjost2s, 2017-07-07:
We have done this already, eg you type new this will automaticly rename the variable to new2, so we have a kind off blacklist here: https://github.com/OpenRoberta/blockly/blob/master/core/reserved_words.js
It just has to be extended.

closed 2017-07-10 by cvejoski