OpenRoberta/robertalab #545: Calliope 2017 programms can't be transfered

stefanscheidt, 2017-07-02:
I can not run a program on a UBS connected Calliope. The lab only shows an error message: 

"Ein Fehler ist beim internen Speichern des Programmes aufgetreten.“

The Calliope shows up as an USB device and I can copy programs I wrote with http://pxt.calliope.cc/index.html. Unfortunately I found no way to export the compiled program out of Roberta Lab.

I use Open Roberta 2.2.6-SNAPSHOT in Google Chrome Version 59.0.3071.115 on macOS 10.12.5 (16F73) and a Calliope mini 2017

Here's the content of Details.txt from my Calliope.

# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 12A0000040304e4500381002000000100000000097969901
HIC ID: 97969901
Auto Reset: 1
Automation allowed: 0
Daplink Mode: Interface
Interface Version: 0242
Bootloader Version: 0242
Git SHA: eab079b8a1326f0a3647e3df33cf5dda13994a1f
Local Mods: 1
USB Interfaces: MSD, CDC, HID
Bootloader CRC: 0x459c23c8
Interface CRC: 0x772b65c8

bjost2s, 2017-07-03:
Please try it again today. Should be fixed

stefanscheidt, 2017-07-03:
Works fine now. Thanks!

closed 2017-07-03 by stefanscheidt