OpenRoberta/robertalab #549: missing distance encoder at preconfigured wait until block (EV3 and NXT)

MaikBaeumer, 2017-07-05:
The preconfigured `wait until` block has a lot of useful sensor values to choose, but many kids like to have an additional `wait until get distance encoder X > xyz` .
Creating this is quite complex in the simple view: Using a preconfigured wait until block, ripping off the whole logic block (because the green sensor block can not be seperated from the logic block) and arrange a new logic block, sensor block and number block.

bjost2s, 2017-07-14:
In many experiments/tutorials are based on the question, how many rotations/degrees should the motors turn to reach a certain distance. That the encoder block offers this already calculated value is not correct, but we cannot remove this anymore because of old programs maybe use this.
But you can create this by yourself: Switch to the expert mode and use the needed blocks. There is no need to rip of anything.

bjost2s, 2017-11-25:
As we are revising all sensor blocks, this feature will be automaticly included, because there is only one description of the sensor modes

bjost2s, 2018-01-11:
solved with #697

closed 2018-01-15 by bjost2s