OpenRoberta/robertalab #550: Sim world editor

Yogibaer2, 2017-07-05:
My first question was how to generate an obstacle on my own sim ground for EV3 or NXT.
The answer was: up to now there is no posibility. That`s the background why I start this issue.

My idea is a complete **sim world development environment.** 
The "start" is a background image. The size should be presetted and the scale to a wallpaper should be defined. Under these conditions parallel to the simulation the background image can be printed or painted to build a real Mindstorms World.

The environment shoud have:
- "obstacel block" (3 dimension size) on which the ultrasonic sensor and the touch sensor can react.
 If wanted a block should be movable (perhaps by a fork lifter) 
- "sound block" on which the microfon can react as well related to the loudness as to a sound (frequencies or melody detection)
- "light block" on which the light- and color-sensors can react (for example: fires at the port of a habour or the flags at ski slalom).
etc. ......
This sounds like a lot of work, but if it is possible to combine roberta-lab with Greenfoot it might be possible to do this with students.

I´m awaiting your comments.
Yogibaer2

bjost2s, 2017-07-07:
Hi, thank you for your ideas.
I'd like to give you some reasons, why the simulation is how it is:
- our simulation is running in a browser, in best case in all browser, also in old versions (not too old), the user doesn't have to configure anything in his or her browser, no installation or activation of plugins, because teachers and students cannot do this - we use only html5 for our simulation. All available 3d libraries are using webGL, to be performend.
- The concept of Roberta and Open Roberta is to learn programming with hands on robots, to motivate the kids. The simulation is only there to give them a first impression of how it would work and that it is useful and motivating to work with a real robot. If we implement all the things you mentioned, there is no need to work with real robots. We could also add to this renewed simulation several robots, so that there is no need to explore on real collaborative robots like in reality.
- The sound block you mentioned makes in my oppion no sence, because non of our allready supported systems has a dedicated sensor that can detect frequencies or melodies.
- The light sensor can already detected colors and lights. It is pointing to the ground, you can upload you own background pictures to use a customized szenario. What you realy want is probably a customizable geometry of the robots, but we want to enable the kids to build there own real robots. Almost none of them (including the teachers) can construct a differential drive robot with lego without instructions. 

So you can see, we do not really need a better simulation, but if you could provide one which can cover all the systems that we are already supporting, that would be great and we would love to set it online. Also feel free to connect it with Greenfoot, if you see an easy way. I think it will be quite hard to connect a modern webinterface with java. 

Hopefully you can understand our mission a bit better now. We are really coming  from real robots and want to inspire young people with this idea and motivation.

ensonic, 2017-07-08:
I think we want to find a way to link uploaded simulation background to accounts, so that they are stored online (and ev. can be shared). Also having a way to define obstacles/walls would be nice (e.g. for a maze).

closed 2017-11-25 by bjost2s