OpenRoberta/robertalab #551: Calliope: Missing Forward Declaration for Functions

MxKy, 2017-07-06:
Short Description
-----------------------
When using functions from the expert tab, no forward declarations are generated for the resulting C++ code. This means that the order of creation restricts which function can call which and this also makes recursion impossible.

Reproduction Steps
--------------------------

  * go to the [OpenRoberta lab](https://lab.open-roberta.org/)
  * create a new project for the Calliope (revision 2016 or 2017)
  * open the expert tab on the left by clicking on the star to show more NEPO blocks
  * create a function `foo()`
  * create a second function  `bar()`
  * let `foo()` call `bar()`
  * try to compile the program by hitting the play-button in the bottom-left

Explanation
----------------

Screenshot of the NEPO-program:
[image]

Resulting source code:

    #define _GNU_SOURCE

    #include "MicroBit.h" 
    #include <array>
    #include <stdlib.h>
    MicroBit uBit;

    int initTime = uBit.systemTime(); 

    void foo() {
      bar();
    }

    void bar() {
    }

    int main() 
    {
        uBit.init();

        release_fiber();
    }

The function `bar()` has not been declared yet, when it is being called inside of `foo()`.
In C / C++ this leads to a compile-time error.

Suggested Fix
--------------------
Add a forward declaration for each function at the top of the compilation unit.
In this example, insert

    void foo();
    void bar();

after the includes.

Dittse, 2017-07-07:
NEPO compiles the placed blocks from top to bottom. With this knowledge, one can position functions in their wished order of declaration. 

The program compiles fine if you position your blocks like this:
[image]

MxKy, 2017-07-07:
@Dittse, thank you for that information. Is this behaviour documented somewhere?

I agree that knowledge of this solves my specific problem, however, it still prevents **recursive** function calls.
If this is by design, I guess everything is fine as it is.

closed 2017-07-17 by cvejoski

elhele, 2017-07-17:
Works fine for Arduino-based robots without any changes. Apparently, the compilation process is different.
However, it breaks for NXT, so it requires a similar correction.