OpenRoberta/robertalab #555: No EV3 Build Instructions

murraycu, 2017-07-10:
The website doesn't seem to have any link to any instructions for building a suitable EV3 model that would be appropriate for the EV3 configuration.

bjost2s, 2017-07-12:
We have removed the direkt links from the lab, because we are supporting many different systems right now, that the list of links in the help menu will be to long. 
You find the instructions still here: https://jira.iais.fraunhofer.de/wiki/display/ORInfo/Set+up+EV3+-+System+leJOS#SetupEV3-SystemleJOS-Buildinginstruction

ensonic, 2017-07-12:
Maybe we can have the link on the robot config screen for the ev3?

murraycu, 2017-07-13:
> You find the instructions still here

Ah, sorry, I now see the drop down menu from the Setup button on the page reached from the "general help" menu item. I don't know why I didn't find that before. So I wouldn't mind closing this bug.

It would be helpful if it said what EV3 kit the instructions need. I think it still requires the Education kit, not the retail kit. At the moment you risk people buying the regular EV3 kit (not even knowing that there is an Education version) specifically to use OpenRoberta.

bjost2s, 2017-07-14:
I think we should offer a second version of the building instruction for the retail version. I will forward this to my colleagues, thank you!

closed 2017-07-14 by bjost2s