OpenRoberta/robertalab #558: Calliope 2017: motor runs with constant velocity

herb91, 2017-07-10:
I expected the motor starts with speed 10% and increase the speed to 80% after a second. But the motor runs with constant speed and stops at the end. With negative values the motor doesn't move.

```javascript
while ( true ) {
  if ( uBit.buttonA.isPressed() ) {
    uBit.soundmotor.motorAOn(10);
    uBit.sleep(1000);
    uBit.soundmotor.motorAOn(80);
    uBit.sleep(1000);
    uBit.soundmotor.motorAOff();
}
uBit.sleep(1);
}
```

cvejoski, 2017-07-11:
There is bug in the calliope firmware that if you use the motors separately they both must be active.
Can you please first set both motors (A and B) to 0 and then try to the rest of the program that you have.

bjost2s, 2017-07-11:
The problem is, that we do not offer a block for using the motor driver with one motor. There is another function in the firmware that can also accept negative values and will work as expected. Our motor block is for using calliope with two motors, for robotic matters.
Maybe we will add the single motor block in the next version.

herb91, 2017-07-11:
@cvejoski 
I've tried this code with the result that the motor A didn't turn.
```javascript
    while ( true ) {
        if ( uBit.buttonA.isPressed() ) {
            uBit.soundmotor.motorAOn(0);
            uBit.soundmotor.motorBOn(0);
            uBit.display.scroll(ManagedString("10"));
            uBit.soundmotor.motorAOn(10);
            uBit.sleep(1000);
            uBit.display.scroll(ManagedString("50"));
            uBit.soundmotor.motorAOn(50);
            uBit.sleep(1000);
            uBit.display.scroll(ManagedString("100"));
            uBit.soundmotor.motorAOn(100);
            uBit.sleep(1000);
            uBit.display.scroll(ManagedString("0"));
            uBit.soundmotor.motorAOff();
        }
        uBit.sleep(1);
    }
```
Without the lines
```javascript
            uBit.soundmotor.motorAOn(0);
            uBit.soundmotor.motorBOn(0);
```
the motor works, but with constant high speed.
I will check the behaviour with 2 motors as mentioned from @bjost2s

bjost2s, 2017-07-14:
Could you please show us how the motor at the example above  is connected?

herb91, 2017-07-15:
In the example the connected pins are:
VM(max 9V): + of a 9V Lego power supply
GND: - of the power supply
Motor A: + of the motor
Motor B: empty
GND: - of the motor

cvejoski, 2017-07-17:
Do you use the public open roberta lab version or local version?

herb91, 2017-07-17:
I have used the open roberta lab version https://lab.open-roberta.org/
With this code (one motor connected to A) 
[image]
the motor starts turning with 100% if the button is pressed until the motor stops.
If the code controls 2 motors
[image]
both motors starts with 40% up to 100% and everything works fine.
This code works for one motor too. It doesn't matter if the motor is connected to A or B.

cvejoski, 2017-07-18:
The solution to the problem will be to implement new block only for one motor. The current motor blocks should be used when one have two motors

cvejoski, 2017-07-31:
check ticket #582 for the new motor block

closed 2017-10-02 by bjost2s