OpenRoberta/robertalab #560: NAO robertalab-naoprogram improvements

VinArt, 2017-07-11:
First thing that we discussed with @bjost2s was that naoprogram should have releases and versions. When hal will be working properly with current code generation (and I think it does, at least almost) we can declare this being version 1.0.0.
Second thing is that naoprogram should be able to kill currently running program, because if the user submits a forever looping program to NAO, the only way to submit another one is to close the program and re-open it and re-connect with the robot. This leads to a bunch of python processes running and consuming CPU time as well as still operating on NAO's hardware. For people it can be confusing, since they might not expect, that their program will run after re-connect. Also currently the only way to get rid of those programs without having shell access is to reboot the robot.
Third, we also discussed that there should be an option to install user's program to autoload after boot. I think this can be achieved by throwing a script to init.d. The problem here is the actual implementation, will NAO ask whether to launch previously submitted program? Should a user put a tick mark in naoprogram so this particular one would be installed to autoload? Yet another problem comes with having no root rights on NAO (at least I did not get sudo with nao user), so the naoprogram is not able to modify system files.

VinArt, 2017-07-11:
Update, there is a file Roberta [0] ~/naoqi/preferences $ cat autoload.ini with following contents:
# autoload.ini
#
# Use this file to list the cross-compiled modules that you wish to load.
# You must specify the full path to the module, python module or program.
[user]
#the/full/path/to/your/liblibraryname.so  # load liblibraryname.so
[python]
#the/full/path/to/your/python_module.py   # load python_module.py
[program]
#the/full/path/to/your/program            # load program

so we could just use that for autoload, I think

bjost2s, 2017-09-17:
what is the current state of this issue?

VinArt, 2017-09-17:
Changes that were in haribo are backported to the main project.

VinArt, 2017-09-20:
44ff7b7 resolves this

closed 2017-09-20 by cvejoski