OpenRoberta/robertalab #565: Code generation variable sequence in assignment

VinArt, 2017-07-14:
For this kind of program the following java code is generated, which will lead to an exception:
[image]

In c/c++ it is not a problem, since variables are declared beforehand and will be just initialised with garbage or 0 depending on the compiler, but for java and python this is a problem.

elhele, 2017-07-14:
Tested on ev3, the code doesn't compile when trying to upload to the robot.

bjost2s, 2017-07-14:
related to #417, please commend on my proposal to  fix the bug in the following way:
"What we could do easily is to make the declaration more save that users cannot change the preassigned blocks, but changing the value. The user than cannot assign a function to the declaration anymore."
Where cannot assign a function means also cannot assign a variable.

cvejoski, 2017-07-17:
I think not allowing to the declaration will be the best solution. In this way we will fix all the problems related to calling sensors that they are not setup yet.

closed 2017-10-02 by bjost2s