OpenRoberta/robertalab #573: NAO speech recognition block

VinArt, 2017-07-18:
Currently speech recognition block from NAO looks like this:
[image]
and generates the following code:
`h.recognizeWord()`
If we take a look at hal.py invocation of this method should also contain vocabulary (before my last commit it was like that too), so the block should accept a list of words or phrases.

Speech recognition on NAO is evenet based, so it might be tricky to use it from blockly (without blocking).

VinArt, 2017-08-04:
Discussed solution would be to start speech recognition engine in the background if this block is used and let it update a lastRecognizedWord variable of hal. So all the speech recognition will be in reading the contents of that variable. Initialisation block would require a vocabulary though.

VinArt, 2017-08-07:
`#!/usr/bin/python

from hal import *
import time

h = Hal()

speechRecognition = SpeechRecognitionModule("speechRecognition")
speechRecognition.setVocabulary(["one", "two", "three"])
speechRecognition.toggleVisualExpression(False)
speechRecognition.toggleAudioExpression(False)

while(True):
    try:
        while True:
            time.sleep(1)
    except KeyboardInterrupt:
        print "Interrupted by user, shutting down"
        speechRecognition.memory.unsubscribeToEvent("WordRecognized", speechRecognition.getName())
        h.myBroker.shutdown()
        sys.exit(0)

`

Speech recognition works like this. I will try and make it better, for now it is like this.

bjost2s, 2017-09-17:
what is the current state?

VinArt, 2017-09-17:
I believe it is done, @cvejoski helped finishing it and it is in sync with haribo

VinArt, 2017-09-20:
Sorry, I meant #560, not #550 above

closed 2017-09-20 by cvejoski