OpenRoberta/robertalab #574: webapp manifest is not expanded during build

ensonic, 2017-07-19:
When building ./OpenRobertaServer/staticResources/manifest.webapp.json we need to expand the project properties.

bjost2s, 2017-07-19:
I do not really know, why this was a bug, but after your commit we have a bug:
[image]
Please always test with deleting the cache

ensonic, 2017-07-19:
The missing icons? I'll take a look.

ensonic, 2017-07-19:
I'll check if maven can do the filtering just for specific files.

ensonic, 2017-07-19:
Now it only expands the expected file types.

bjost2s, 2017-07-19:
As I do not know exactly what the reason for your changes are, I think should leave this open until someone, you?, has done more tests.

ensonic, 2017-07-20:
Sorry I don't understand. Before this change 	da5fc63 the manifest.webapp.json was not expanded during the build, now it is.

``` patch
diff ./staticResources/manifest.webapp.json ./target/classes/staticResources/manifest.webapp.json
4c4
<   "version": "${openRobertaServer.version}",
---
>   "version": "2.2.7",
8c8
<     "url": "${project.parent.url}"
---
>     "url": "http://www.open-roberta.org"
```

I compared the whole ./target/classes/staticResources with the source dir and no other files are touched, as explained in the commit message.

bjost2s, 2017-07-20:
ah, ok.
Wouldn't it be easier, more safe, to have a dedicated filter only for this one file. Otherwise we have to take care of this each time we add new files/binaries.

ensonic, 2017-07-21:
I can do this as well. I thought it is a good idea to have the filetype list in addition to protect those file types in general.

commit 14b78f33 (2017-07-19): Expand the properties in the webapp manifest.

These properties come from the pom.xml. For the version switch to the human readable
server version.

Fixes #574
commit da5fc636 (2017-07-19): Expand the properties in the webapp manifest.

These properties come from the pom.xml. This change registers binary extensions
we use for files under staticResources so that those are left unchanged.

Fixes #574
closed 2017-07-22 by ensonic