OpenRoberta/robertalab #580: when having a tab oopen (code, help, sim) and resizing the window the tab-handles get detached

ensonic, 2017-07-28:
below I opened the code view, then grabbed the right window border and made the window smaller. If I'd make the window larger, the '<>' icon would be moved under the code pane. The issue also happens in develop.

[image]

VinArt, 2017-07-31:
Interesting, are there special conditions to reproduce? For me strange things happen only if I resize the window (Linux, Chrome 60, X11), but when I start dragging the pane the '<>' icon goes back where it should be.

ensonic, 2017-07-31:
Same here. Tested with chrome/firefox on linux. And yes, it only happens when you resize the width of the window.

VinArt, 2017-07-31:
Also this is true for all other buttons, e.g. sim. So must be a common button bug.

elhele, 2017-07-31:
Another button bug:
if in the end of the welcome tour I don't press the simulation button, but just close the tour, I cannot use the simulation button and cannot close the simulation without reloading the page. Ev3's simulation from the tour will just stay there even if I choose some robot that doesn't have any simulation.

closed 2017-09-12 by cvejoski