OpenRoberta/robertalab #585: Feature request "pin pressed" at get-sensor-value in Calliope

alexgit2k, 2017-08-06:
Option "pin pressed" should be available in the get-sensor-value-block:
[image]

VinArt, 2017-08-11:
Maybe this should look like: wait until pin high/pin low?

alexgit2k, 2017-08-13:
Ah, I forgot that the pins can not only be touched, but also can have a digital or an analog value. So the new options should be:
- get "pin (touched)" 0/**1**/2/3 **=**/≠ **true**/false
- get "pin (digital)" 0/**1**/2/3 **=**/≠ **true**/false
- get "pin (analog)" 0/**1**/2/3 **>**/≥/</≤/=/≠ Integer (**50**)

Bold-values should be default values.
Touched and Digital could be grouped together as "pin (touched/digital)".

closed 2017-10-02 by bjost2s