OpenRoberta/robertalab #586: Feature request: random numbers for Beginners

alexgit2k, 2017-08-08:
Random numbers are part of beginner programms like dice, rock–paper–scissors, oracle, ... but in NEPO it is only available for Experts. Therefore the random integer block should be available for Beginners.

[image]

bjost2s, 2017-08-28:
We discussed this and I think we will give it a try. The problem was, that we 
- want  to keep the beginner toolbox small
- beginners, exspecially 3. class kids have no idea, what a random number really is.
- usually we keep the lab consistent, that means we would have to change all beginner toolboxes from the different systems, that we support. But for our robots, like EV3 or NXT, nobody has asked for this because it is not used in beginner mode.
I personally would suggest to use sensor values to make decisions and not random values.

alexgit2k, 2017-08-31:
Thanks. Using sensor values for random numbers are no option, they are not random and the range has to be transformed to the required range which requires much more knowledge than 3rd grade.
Unterstanding random numbers was no problem, randomness is even part of 1st class.

closed 2017-09-18 by cvejoski