OpenRoberta/robertalab #587: Unable to move sensor block out of "wait until" block

alexgit2k, 2017-08-08:
In the control-block "wait until" the sensor-block ("get button (pressed) A") can not be moved out only the complete logic-block and the small logic-block ("true"):

[image]

bjost2s, 2017-08-28:
This is wanted, because these three blocks are internally connected. Usually blocks are undependent, but here, we have to change the surrounding logic block, if the inner sensor block changes, eg the type.

alexgit2k, 2017-08-31:
Can you visualize this behaviour by e.g. using the control-block-color for the logic-block and removing the borders? It is currently not understandable why the block can not be moved and I have seen several unsuccessful tries.

ensonic, 2017-08-31:
Just to mention. I was not aware of this and always thought it is a blockly bug :/ I think it is unexpected because you can manually assemble the logic block and then also disassemble it again. When using a logic block that you manually assemble, of course you need to remove both blocks before you can change the type.

bjost2s, 2017-11-25:
Think we can make it movable and offer this block also in the toolbox. We will see how it works ;-)

bjost2s, 2018-01-11:
fixed with #697

closed 2018-01-15 by bjost2s