OpenRoberta/robertalab #588: List length bug

Dittse, 2017-08-09:
While I was trying to program a Fortuneteller with the Calliope mini, I recognized strange behavior of lists.

The whole Program worked fine until the 43st Element in my list of fortunes was added. With 43 fortunes, the Program does nothing on the physical Calliope.  I tested a list of strings, which only consisted of numbers, and it worked fine. If you try to initialize a second list, it will automatically show the list error as If you tried to get the -1st or (last + 1)st Element.
[FortuneTeller2.zip](https://github.com/OpenRoberta/robertalab/files/1211614/FortuneTeller2.zip)

cvejoski, 2017-08-10:
Too many fortunes :). The problem is that calliope does not have much memory and the size of the content of the list of fortunes is to big for Calliope

Dittse, 2017-08-11:
How can that be if I already wrote Programs that needed more than double the memory of the Fortuneteller?
They work fine but just don’t contain lists.

VinArt, 2017-08-17:
@Dittse Could you post an example of such a program?

cvejoski, 2017-08-20:
The problem is that by default the bluetooth  module is on and that takes a lot of memory. Maybe we should turn it off and make a block that explicitly turns on the bluetooth

bjost2s, 2017-08-28:
I think a block that turns bluetooth on is a bit difficult to understand for user, maybe we should have a configuration for calliope to enable users to add bluetooth to the program.

closed 2017-11-25 by bjost2s