OpenRoberta/robertalab #594: lab.open-roberta.org shows "Unfortunately you can not use this block in the simulation!" but it works in simulation

corvus-albus, 2017-08-23:
Hello!
I switched from enchanting/NXTonStage to openroberta and I'm really excited to have simulation and export to the robot at one place. As I ported a little programm I found two minor bugs:

I wrote a program with "Alle meine Entchen" for my beginner-robotic-club. I use NXT-simulation (and real NXT).
As I open the simulation the function-block "n2F" displays an attention-sign. Clicking on it it shows the above mentioned comment ("Unfortunately you can not use this block in the simulation!"). But the function works fine in simulation.
In addition the source code does not show up as I click on "< >" ("open/close source code view"). 

The program is attached. (fileextension changed from xml to txt - you cannot upload xml-files)
[NEPOprog.txt](https://github.com/OpenRoberta/robertalab/files/1244589/NEPOprog.txt)

Thanks for the good work.

Cheers 
corvus-albus

corvus-albus, 2017-08-24:
I tested a little bit further. 
the fact that the source code is not shown for the attached programm comes with the block
"in list find first occurence of ". 
I searched in NXC for such a command and did not find anything. It might be that there is no possibility to translate this block to NXC. If I'm right, it would be the best to drop this command-block for programming with NXT and do not show it at the flyout.

Cheers 
corvus-albus

bjost2s, 2017-08-24:
We had some problems to implement list functions in nxc, because they are not supported in the standard firmware. We will check what could be wrong and you are absolutely right, if we can not support it, we will remove it from the toolbox

elhele, 2017-08-25:
Hello, @corvus-albus , there is such a command in our implementation, but it only works with variables, not when you insert the array directly. See the modified program attached, it worked for me:
[NEPOprog_ArraysToVars.txt](https://github.com/OpenRoberta/robertalab/files/1251436/NEPOprog_ArraysToVars.txt)
But you are right, we should add the possibility of inserting arrays directly.

corvus-albus, 2017-08-26:
Hello elhele,
it works, thank you. 
I had to remmove the  ' tags=""' entry at the beginning of the file to load it, but afterwards everything was fine.
It is not nice to have such a long start-block. Perhaps it would be a good idea to add the possibility to collapse the list declaration.

To go back to the comment I started the issue first. I think the comment should be: 
"Unfortunately you can not use this block only in the simulation!" 
As I experienced problems it was not with the simulation but with the fact that it did not compile or gave strange effects on the brick.

Thanks for the hints.
corvus-albus

bjost2s, 2017-09-28:
what is the current state, resolved, won't fix, postponed?

elhele, 2017-09-29:
@bjost2s I'd rather say "won't fix" with the current NXC that we use. It is not too hard to implement it for the program like this, but requires too much work to implement it for all possible cases. It would be easier to create a message that would advice to use arrays as a variable, not directly.

closed 2018-11-16 by VinArt