OpenRoberta/robertalab #595: Feature request: add a "play note"-block like it is in scratch

corvus-albus, 2017-08-23:
Hello!

Pupils love to see the NXT dancing. But the "play frequency"-block is hard to use: to program a melody you must know the frequencies of the notes to play.
It would be very nice if you could add a "play note"-block with a little keyboard integrated like there is in scratch.

Keep up the good work
corvus-albus

bjost2s, 2017-08-24:
Please check the calliope system in the lab. There we already have such a kind of block. Do you mean this?

corvus-albus, 2017-08-25:
Yes, that's it. I would love it if the keyboard would be available for NXT too!

Cheers 
corvus-albus

corvus-albus, 2017-08-26:
Sorry, closed the issue unintentionally.

bjost2s, 2017-08-28:
you can look at the source code in e.g. calliope view, when you used one of the note blocks. I made it simple, a whole note has a duration of 2000 ms, a half note 1000 ms ....
I will check, if we can make it for the next release.

bjost2s, 2017-10-03:
new issue for all systems here: #640

closed 2017-10-03 by bjost2s