OpenRoberta/robertalab #596: Feature request: add hidden local variables in functions

corvus-albus, 2017-08-23:
Hello!

It would be very useful to have "hidden" variables in functions. If you like to add for example a loop variable in a function there will be a useless input position at the function-block. In effect I see no possibility to create a non-input local variable in a function. 

I think it would be very useful to have those hidden local variables in functions.

Keep up the good work.
corvus-albus

bjost2s, 2017-08-29:
for those variables we need a place to define them. The only fixed block in functions is the function itself, but here we already define parameters. If we would have a declaration block it will be difficult to check where they are placed, e.g. after the first use, or in another function...
Any ideas?

corvus-albus, 2017-08-30:
I first was a bit confused because in the start block the variables are called "variables", the parameters in the function block are called "varables" as well. I remember I had a look at the wiki page to understand the situation. 
I suggest to name the parameters "parameters" in the function block even if there is no further change.

I have just two idea:
1. You could add a dropdown field in the function block, where you could choose between "parameter" and "local variable". But I admit that I'm not sure if this isn't to confusing. 
2. You could add a second function block, that is styled like "if-then-else"-block with a "parameter" part and a "local variable" part. That would not be that convenient. You would have to copy all blocks to another function block in the case you realize the need of a local variable. But it is less confusing.

Cheers
corvus-albus