OpenRoberta/robertalab #598: NXT - division of 1:2 gives 0

corvus-albus, 2017-08-26:
Sorry for opening the next issue but I'm testing quite a bit.

If you calculate 1:2 in the simulation you get as result 0.5 as expected. 
On the NXT-brick you get 0 . It seems that there is a int-division with an int-result.
If you divide 1:(2^1) you get 0.5 on the NXT as well. 

The results are right if one of the numbers is not an integer. 
For example: 
  3:4 gives 0 
  1.5:2 gives 0.75

I would expect the division to be a float-division. To get the int-result I would use the round-block.

Cheers 
corvus-albus

[NEPOprog_NXT-division-examples.txt](https://github.com/OpenRoberta/robertalab/files/1253530/NEPOprog_NXT-division-examples.txt)

VinArt, 2017-08-28:
I believe this is done because there are no dedicated div and mod operators, so 1 / 2 would be integer division and 1.0 / 2.0 would be float division, so as a workaround you can ad .0 to your number. We will think of a way to resolve this issue. Thanks you for the information.

bjost2s, 2017-08-28:
We decided here #60, that devisions (and all other math) will happen with floats. So this is a bug and we will fix it.
Thank you

closed 2017-09-11 by cvejoski