OpenRoberta/robertalab #603: Feature request: flashing multiple robots simultaneously

VinArt, 2017-09-05:
Create Agent integration #602 #492 allows users to see a list of multiple robots if more than one robot of the same type is connected. As a user I may have up multiple robots that I want to pre-flash with demonstration programs or use the same program on all robots that implements some robot interaction behaviour. For these purposes it would be nice to have not only "play" ('>') button, but also 'play all' ('>>') button that will flash this program to all robots in the list.

bjost2s, 2017-09-07:
Unfortunately we only have one robot system, where the create agent works. Additionally most users have the problem of not having any robot system so having more than one is probably rare.

bjost2s, 2018-04-12:
Please reopen, if this is still a whish

closed 2018-04-12 by bjost2s