OpenRoberta/robertalab #604: Feature Request: Simulator Data Logging

psychemedia, 2017-09-07:
One of the ways we motivate learners to write programmes is to get them to write programmes to use the robot as a mobile data logger.

Is there any way of logging and either displaying or downloading logged sensor data in the simulator? For example,  drivingt he robot over the coloured lines in the simulator and then displaying the light sensor data either against either time or wheel rotations?

(I am particularly keen on being able to do this *in the simulator*.)

VinArt, 2017-09-07:
First, thank you for the idea, we shall discuss it. Second, there is a small hack that you can do to log sensor data through your browsers console (on Chrome for example one can open it with F12 button).
The idea is simple, just make an timer that will print out timestamp and sensor data to console.
> function logToConsole() {console.log(Date.now(), $("#valuesContent:contains('FPS')").text());}
// we create a function that we'll call periodically.
> myTimer = setInterval(logToConsole, 500);
// we setup a recurring event, that will call the above function each 500 ms
> clearInterval(myTimer)
// we kill this event when we want to stop logging.

the result of the above manipulations will look something like this:
1504789670648 "FPS63Time0sRobot X224Robot Y231Robot θ0°Motor left0°Motor right0°Touch Sensor0Light Sensor33%Ultra Sensor114cmColor Sensor "
1504789671148 "FPS56Time0sRobot X224Robot Y231Robot θ0°Motor left0°Motor right0°Touch Sensor0Light Sensor33%Ultra Sensor114cmColor Sensor "

ensonic, 2017-09-07:
While we can technically do it for all simulated robots, it is probably surprising (or disappointing) to some people if it will not work with the actual hardware. And for the physical robots (which is our focus) this feature requires:
1. some connection of the robot and the computer for live data capture
1. some log buffer on the robot to look at the data once the robot is reconnected

In addition the programs need to be augumented to capture and we need to extend the APIs to exchange the log data. So it is quite a bit of work. 

That being said it would be a cool feature for sure :)

psychemedia, 2017-09-07:
Thanks for commenting back - and suggesting the hack:-)

We're looking to replace our own simple simulator ([OU-RobotLab](https://blog.ouseful.info/2017/08/14/replacing-robotlab/)) which was originally designed as a piece of educational software to work with Lego RCX2 and as a standalone simulator, so I appreciate the desire to have functional equivalence between simulator and real robots.

FWIW, I made a quick note of some of the features we had in RobotLab that don't appear in RobertaLab:

- step trace to step through programme commands one at a time;
- data logging and data log charting;
- custom audio files (we had the robot counting aloud!);
- simulator pen down (so we could look back at the path followed by the robot).

bjost2s, 2017-09-07:
Hi and thank you for your suggestions.
for the simulator pen down, please use the drawing background in our simulation, it is the second one.
for step tracing it would be interesting for us, how this will work with non blocking blocks, like drive forward.

psychemedia, 2017-09-08:
@bjost2s Re: pen down - thanks for that tip.

Re: non-blocking blocks - the RobotLab simulator highlighted each blog when it was the focus of the control thread.

One of the ways we use to explain our (RCX2 targetted) programs to students was that the motor power/on/off commands actually defined state in a motor state table which defined the motor actions.

So a fuller trace would actually be:

- command currently being processed
- motor state table
- sensor state table