OpenRoberta/robertalab #612: Feature request ev3-dev: support NXT Light Sensor

corvus-albus, 2017-09-12:
Hello,

ev3dev does support NXT light sensors according to their documentation: [click](http://docs.ev3dev.org/projects/lego-linux-drivers/en/ev3dev-jessie/sensor_data.html#lego-nxt-light-sensor). It would be very nice to support this device in Open Roberta. As the displays of our NXTs at school slowly give up (up to now 50%) the sensors are still alive. It would be very nice to have the possibility to use them with the EV3.

Cheers 
corvus-albus

bjost2s, 2017-10-09:
We always have to check new features for ev3 against all systems. We will have a look into the lejos libraries and come back to this feature ASAP

corvus-albus, 2017-10-10:
Thank you for the new snapshot 2.3.1 and for your promising reply. I had a look at the lejos EV3 api documentation and found a description:
http://www.lejos.org/ev3/docs/lejos/hardware/sensor/NXTLightSensor.html
I hope that is helpful.

Cheers
corvus-albus

bjost2s, 2017-11-22:
Thanks, we will add it to our wish list here #458

bjost2s, 2018-02-07:
see #458

rbudde, 2019-06-26:
see #458

closed 2019-06-26 by rbudde