OpenRoberta/robertalab #623: Icon for create agent tray

VinArt, 2017-09-25:
Create agent has a tray menu that currently uses arduino icon. This can be replaced with OpenRoberta logo if needed.

VinArt, 2017-09-25:
I am using the icon from staticResources/css/img of the OpenRobertaServer https://github.com/OpenRoberta/arduino-create-agent/commit/1ff9e899c87b88da4bcb03ee5cd1c1cdb70b84dc. However, MacOS had a different approach, having a black-on-white icon for regular theme and light version for,  I guess, light theme. Do we provide theme-depended icons for MacOS?

bjost2s, 2017-09-25:
The new Open Roberta Layout does not provide a black-on-white icon, so because of our CI it is not allowed to create one. Please do not create own versions, but ask for needed versions of the icons

bjost2s, 2017-09-28:
What icons are needed finally, please provide more information, size, type, ....

VinArt, 2017-09-29:
So, we need tray icons for Windows, Linux, MacOS.
Windows and Linux icons can be the same. There should exist 2 versions, for active and idle states of create agent. Active state is depicted by a coloured icon, whereas idle state is depicted by a grayscale icon. Size of original icons for Linux and Windows were 49 by 49 pixels. They can be observed here: https://github.com/arduino/arduino-create-agent/tree/devel/icon files icon_linux.png icon_hiber.png
My tray (Linux KDE 1366x768 screen resolution at 15.4") is 29 pixels high.
My tray (Windows 10 1366x768 screen resolution at 15.4") is 40 pixels high.
Regarding this fact I think that for non fullhd or 4k monitors 49 pixels are sufficient. 

File format should be png, since later these icons are transformed to Go data arrays anyway.
MacOS has another set of icons for active state, they can be found under the same link, files icon_mac.on icon_mac_lite.png (MacOS theme depended) a gray-scale version is also needed for idle state.

Icons of bigger size are needed for menu entries in start menu in all systems. I belive 256x256 is sufficient for Linux and Windows. MacOS probably requires higher resolution due to higher screen DPI

On Windows tray icons look like this:
active:
[image]
idle:
[image]

On Linux it depends on the DE being used, but the idea is the same.

It would be nice to utilise NEPO.ico as found here:
https://github.com/OpenRoberta/robertalab/tree/master/OpenRobertaServer/staticResources/css/img file NEPO.ico

To sum up:
1 coloured icon (N logo) 49x49 pixels, png (Linux/Windows)
1 grayscale icon (N logo) 49x49 pixels, png (Linux/Windows)
1 black-and-white icon (N logo) original resolution was 38x38, png, (MacOS)
1 lite icon (N logo) original resolution was 38x38, png, (MacOS)
1 coloured icon (N logo) 256x256, png (Linux/Windows)
1 .icns icon pack (N logo) multiple resolutions, starting 256x256, icns (MacOS)

bjost2s, 2017-10-08:
icons and images are in the repo now

closed 2017-10-09 by bjost2s